fotisk07/SI-Project #29: Breaking down the measurement set

CeType, 2019-10-29:
- As of now, the simulator submits all of the data to LiMap as one big list. That is unrealistic and doesn't allow to test the ability of LiMap to handle repeated smaller chunks of data rather than all in one

_Please reference this issue in commits or pull requests aiming to implement/solve the discussed topic so as to close this issue when merged into master._

fotisk07, 2019-10-29:
I mean  I could just wait for you to complete a "tour" and then you sent me a big list with stuff

CeType, 2019-10-29:
Why not, but I was thinking of it with #30 and the future IRL transposition in mind, where it'd be really short (depending on the graphics redraw time and computation)

closed 2019-11-10 by fotisk07