fotisk07/SI-Project #30: Animating the LiDAR result plots

CeType, 2019-10-29:
- Show the progression in time of the LiDAR's measurements as it processes more and more data points. Could benefit from #29 by giving a realistic way to redraw the graphics

fotisk07, 2019-10-29:
I'm working on it

CeType, 2019-10-29:
I know, it's for organisation purposes, according to GitHub's guides, that's a good way of managing future features

CeType, 2019-10-29:
The way I'd see it (and that I think would integrate well in the IRL port of LiMap) is maybe something with a loop (IRL LiMap in [])

->Get data from LiSim[Get data from serial buffer] 
-> Run the mapping with the previous map as starting point and the new data
-> Redraw graphics
-> loop

But that's only a suggestion, you're free to do whatever you want with it (use it fully, partially, discard it fully)

fotisk07, 2019-10-29:
So yeah I guess it's gonna work pretty much the way you imagined it

CeType, 2019-10-29:
Couldn’t that be problematic for the IRL version ? I mean there must be a pretty heavy I/O overhead from using a file as an exchange thingy

CeType, 2019-10-29:
Plus possibly overhead from ensuring data integrity

fotisk07, 2019-10-29:
maybe but I don't see a better way especially with those shitty libraries

CeType, 2019-10-29:
Can you publish your working branch so I can tekkaluk ?

fotisk07, 2019-10-29:
Sure

fotisk07, 2019-10-29:
It's not finished and i'm taking a break rn so it's tagnled but you can have a go

CeType, 2019-10-29:
Oki

CeType, 2019-10-29:
I'm completely unable to work on this, Mac OS refuses to draw any dynamic PyPlot thingies, I have tried one of the bazillions of semi-solutions on SO but none worked. This seems to be a recurring problem sooo... I'll try to look more into it later

CeType, 2019-10-29:
I can work on CLI stuff tho

fotisk07, 2019-10-29:
I'm telling you dude I HATE ANIMATION

fotisk07, 2019-10-29:
I think I will maybe have a working solution by tonight (depends and when Madison will be back 😉

fotisk07, 2019-10-29:
That's good i was thinking of doing that sometime. Use argparse it's pretty straighforward

CeType, 2019-10-29:
Yep I know

fotisk07, 2019-10-29:
After some CONSIDERABLE AMOUNT OF WORK, I have an animation up and running!

fotisk07, 2019-10-29:
There are some bugs, but i don't think they are due to the animation file

CeType, 2019-10-29:
I bet it simply won’t work on my machine... I’ll try to find a Mac OS specific workaround

closed 2019-11-10 by fotisk07