fotisk07/SI-Project #43: Animation Black and white, small, and infinite loop

CeType, 2019-11-11:
Trying to enable the -ani option results in a very small window (about 1.5 cm^2) and the plot being black and white. Besides, there is no way to stop the program other than killing the process.

fotisk07, 2019-11-11:
It's normal, the animation displays an image and we are feeding in a 2d array with values from 0 to 255 so the image is bound to be black and white

fotisk07, 2019-11-11:
To stop the program you press q a couple of times

fotisk07, 2019-11-11:
And the animation is small because the array is small if you use bigger dim you get a bigger picture, if I scale it it becames uglu and not ncie

fotisk07, 2019-11-11:
Latest commit makes the matplotlib plot nice and blue as they were before

fotisk07, 2019-11-11:
I can write a colormap function that changes stuffs and makes the animation prettier i guess but not sure if the black and white is a huge nuissance

fotisk07, 2019-11-11:
Last commit made the animation colorful, a special treat just for you!

fotisk07, 2019-11-11:
Should I close the issue then?

CeType, 2019-11-11:
Yay

closed 2019-11-11 by CeType